stm32-rs/stm32-rs #1213: USB OTG vs USB OTG HS

usbalbin, 2025-05-02:
When trying to update the pac in [H7-hal's pac staging branch](https://github.com/stm32-rs/stm32h7xx-hal/tree/pac16-staging) from v0.16 to v0.17 i ran into some problems. H7-hal's [rcc code](https://github.com/stm32-rs/stm32h7xx-hal/blob/pac16-staging/src/rcc/rec.rs#L498) seems to assume a lot of usb related fields share the same prefix [usb1otg](https://stm32-rs.github.io/stm32-rs/STM32H743.html#RCC:AHB1RSTR:USB1OTGRST), however somewhere between stm32h7-staging v0.16 and v0.17 some fields got [usb1otg**hs**](https://stm32-rs.github.io/stm32-rs/STM32H743.html#RCC:AHB1LPENR:USB1OTGHSULPILPEN) as a prefix instead. Is this intended or should this be fixed?

burrbull, 2025-05-02:
Not sure.
But this should be unified.

usbalbin, 2025-05-02:
Also I can not seem to find the [FSDEF](https://stm32-rs.github.io/stm32-rs/STM32H743.html#SAI4:FRCR%20[A]:FSDEF) field even when using `stm32h7 = { git = "https://github.com/stm32-rs/stm32-rs-nightlies" }` eventhough. I must be missing something since we can see it in the HTML table.

```
error[E0599]: no method named `fsdef` found for mutable reference `&mut W<FRCRrs>` in the curre
nt scope
   --> src/sai/pdm.rs:296:34
    |
291 | | /                       audio_ch_a.frcr().modify(|_, w| {
292 | |                             w.fsoff()
293 | |                                 .clear_bit()
294 | |                                 .fspol()
295 | |                                 .set_bit() // FS active high
296 | |                                 .fsdef()
    | |                                 -^^^^^ method not found in `&mut W<FRCRrs>`
    | |_________________________________|
    |
```

usbalbin, 2025-05-02:
[FSDEF](https://stm32-rs.github.io/stm32-rs/STM32H743.html#SAI4:FRCR%20%5BA%5D:FSDEF) seems to be marked as `read-only`. [The table 51.6.6 of rm0433](https://www.st.com/resource/en/reference_manual/rm0433-stm32h742-stm32h743753-and-stm32h750-value-line-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) seems to indicate readonly while the description seems to indicate otherwise

> Bit 16 FSDEF: Frame synchronization definition.
> **This bit is set and cleared by software.**
> 0: FS signal is a start frame signal
> 1: FS signal is a start of frame signal + channel side identification
> When the bit is set, the number of slots defined in the SAI_xSLOTR register has to be even. It means
> that half of this number of slots is dedicated to the left channel and the other slots for the right
> channel (e.g: this bit has to be set for I2S or MSB/LSB-justified protocols...).
> This bit is meaningless and is not used in AC’97 or SPDIF audio block configuration. It must be
> configured when the audio block is disabled

usbalbin, 2025-05-02:
#749 #752 again?

[Here](https://github.com/stm32-rs/stm32-rs-nightlies/blob/31571182aeb1d65616381664624f9249ee365463/stm32h7/src/stm32h743v/sai4/ch/frcr.rs#L140)(read) , but not [here](https://github.com/stm32-rs/stm32-rs-nightlies/blob/master/stm32h7/src/stm32h743v/sai4/ch/frcr.rs#L176)(write)

closed 2025-05-04 by usbalbin