stm32-rs/stm32-rs #1245: FYI for new warn-by-default lint `mismatched_lifetime_syntaxes`

jieyouxu, 2025-06-08:
Hi stm32 maintainers, we landed a [new warn-by-default lint `mismatched_lifetime_syntaxes`](https://github.com/rust-lang/rust/pull/138677) on the nightly channel. The lint is intended to warn on cases where the *same lifetime* is referred to by *different syntactical forms* between function arguments and return types, which has been a long-standing source of confusion for users. See <https://doc.rust-lang.org/nightly/rustc/lints/listing/warn-by-default.html#mismatched-lifetime-syntaxes>.

For the stm32 crates, it was reported that the performance of the lint might be a concern, particularly as the non-`hal` crates are generated. We [analyzed the performance regression](https://rust-lang.zulipchat.com/#narrow/channel/131828-t-compiler/topic/Perf.20regression.20for.20warn-by-default.20lint/with/522890574), and found that the performance regression is dominated by the emission of *a lot* of warnings on the generated code.

**This is expected to only impact build times when working on the generated crates, not when the crates are used as dependencies because in that case they will be built with `--cap-lints=allow` preventing warnings from being emitted.**

To address the build time impact for when working on the generated crates:

- You may wish to `#[allow(mismatched_lifetime_syntaxes)]` on the generated crates. Or,
- You may wish to adjust the generation implementation to emit code that uses consistent lifetime syntaxes for function arguments and return values.

For your reference,

- See [performance regression discussion thread](https://rust-lang.zulipchat.com/#narrow/channel/131828-t-compiler/topic/Perf.20regression.20for.20warn-by-default.20lint/with/522890574).
- See [feedback on the `mismatched_lifetime_syntaxes` thread](https://rust-lang.zulipchat.com/#narrow/channel/213817-t-lang/topic/Feedback.20on.20mismatched_lifetime_syntaxes/with/522920341).
- Or you can also open new issues on [rust-lang/rust](https://github.com/rust-lang/rust) and tag the issue with `@rustbot label: +L-mismatched_lifetime_syntaxes`.

(Please use the threads or new rust-lang/rust issues for feedback, as otherwise I may lose the notification.)

jieyouxu, 2025-06-08:
I'll open a (copy of) this issue over at the svd2rust generator repo as well, unless you are able to move it if you have permissions.

jieyouxu, 2025-06-08:
> I have write permissions in both repos, but I don't have such option. Maybe it is because they are under different "organization"s (`stm-32` and `rust-embedded`).

Apparently the limitation is sth like

> You can only transfer issues between repositories owned by the same user or organization account.

Let me just open an copy of this issue.

therealprof, 2025-06-08:
@jieyouxu Thanks lot for following up so extensively.